IEAWindSystems/IEA-15-240-RWT #257: regarding the issue of calculating blade natural frequencies

vacupenda, 2025-04-08:
Hello,
First, in the IEA-15-240-RWT_AeroDyn15_blade.dat file, I would like to understand the meaning of the parameters circled in Figure 1 (highlighted in the image).
[image]

Second, when attempting to calculate blade natural frequencies using Bemodes, several parameters need to be specified (marked with red circles in Figure 2).I don't know where to find these parameters, or how to use OpenFAST to calculate the natural frequencies of a stationary blade using free-decay oscillation. I'm unsure about the specific steps to perform this operation.
tw_iner (inertial principal axis angle)
cg_offst (center of gravity offset)
sc_offst (shear center offset)
tc_offst (tension center offset)
[image]

ptrbortolotti, 2025-04-08:
hi @vacupenda, the inputs from your first question are only relevant to water power turbines, so you can simply ignore them

we don't really use BModes anymore. if you are interested in the natural frequencies of the stand-still blade you could simply extract them from the wisdem outputs, these are `rotorse.rs.frame.flap_mode_freqs` and `rotorse.rs.frame.edge_mode_freqs`. alternatively, you can linearize your openfast model (look for the linearization flag in the .fst file) and then process the .lin files with these scripts https://github.com/OpenFAST/openfast_toolbox/blob/main/openfast_toolbox/linearization/examples/ex3a_MultiLinFile_Campbell.py

this second option is a little more elaborate and will require some expertise

vacupenda, 2025-04-08:
Could you please explain how to specifically perform the extraction in WISDEM as mentioned in the first method?

ptrbortolotti, 2025-04-08:
if you just run wisdem, in your output folder you will see several files. just open the .csv output file and look for `rotorse.rs.frame.flap_mode_freqs`

let us know if this works

closed 2025-05-13 by ptrbortolotti