IEAWindSystems/IEA-15-240-RWT #258: Unable to find the corresponding projection target element for node 50.

vacupenda, 2025-04-09:
When I was modifying 'beandyn.dat' and generating the corresponding 'Aerodyn.dat', then running OpenFAST, the following error occurred. How should I resolve this error?
Below are the details of the 50th node in Aerodyn.dat and BeamDyn.dat, along with the error message from OpenFAST.
[image]

[image]

[image]

ptrbortolotti, 2025-04-09:
hi @vacupenda, this starts being an OpenFAST related question more than a question for this repository. I don't have great suggestions for you, I've run into the same issue in the past myself. my recommendation is to confirm that you are specifying the correct 3D coordinates of the aerodynamic centers in AD15 as well as the correct 3D coordinates of the structural line in BD. even small deviations will not be acceptable in OpenFAST. for example, your tips do not seem to align nicely along the z axis

gbarter, 2025-04-09:
Not sure, but this may be related to #255 too

gbarter, 2025-04-09:
I was not able to recreate your error using the latest model files.  Hopefully @ptrbortolotti 's suggestions will help.

closed 2025-05-13 by ptrbortolotti