IEAWindSystems/IEA-15-240-RWT #260: The issue of output power in a 15MW wind turbine blade

vacupenda, 2025-04-14:
The rated wind speed for the 15MW wind turbine blade is 10.59 m/s. However, when I used OpenFAST with BeamDyn to calculate the steady wind condition at 10.59 m/s, the output power was only 12MW. Why does it not reach the rated output power of 15MW at this wind speed? I have disabled the degrees of freedom for both tower hydrodynamics and mooring dynamics.

gbarter, 2025-04-14:
Please see #242 to ensure it is not a repeat issue.

The ~10.6 m/s for rated speed is taken from WISDEM's idealized steady state simulation.  Exact rated power seen in OpenFAST will depend on model settings.  I would expect it to be a bit higher than 12MW though. . .

vacupenda, 2025-04-15:
Yes you are right ,it is a bit higher than 12MW  and when wind speed is closed to 12m/s  it would reach the rated output power of 15MW.

closed 2025-05-13 by ptrbortolotti