IEAWindSystems/IEA-15-240-RWT #261: Natural frequencies of IEA-15-240-RWT-UMaineSemi (tower and blade)

Dellavar, 2025-04-28:
Hello,

I am currently doing the frequency analysis/identification of natural frequencies of the IEA-15-240-RWT-UMaineSemi floating machine using OpenFast Linearization process. I am using the script runCampbell.m from matlab-toolbox github repo (using octave to run the script). With the 6 platform DOFs enabled ,4 tower DOFs enabled, all other DOFs disabled and using CompElast=1, CompInflow=0 , CompHydro=1,CompMooring=3 (CompAero=0,CompServo=o,CompSub=0,CompIce=0,MHK=0), I am getting the Natural (undamped) frequency (Hz) of ED 1st tower fore-aft bending mode DOF as 0.502257Hz and ED 1st tower side-to-side bending mode DOF = 0.508558Hz. Thge operating point is Rotor speed=0, wind seed =0. In Table 8. Floating Tower Properties of “Definition of the UMaine VolturnUS-S Reference Platform Developed for the IEA Wind 15 Megawatt Offshore Reference Wind Turbine Technical Report” IEA Wind TCP Task 37 (July 2020), it is specified that 1st FA bending mode is 0.496Hz and 1st SS bending mode is 0.483 Hz.
Can you please tell me if this difference in tower frequencies is accepted, or I might be doing something wrong in the linearization process? Are the frequencies given in the definition document the coupled frequencies? 

Also, in Definition of the IEA Wind 15-Megawatt Offshore Reference Wind Turbine, the blade frequencies are given as First flapwise natural frequency 0.555Hz and First edgewise natural frequency 0.642 Hz. Is this the uncoupled frequency of the blade? When I run runcampbell.m with only the blade DOFs enabled in the model (all other DOFs disabled and enabling only CompElast to 1 in .fst (and all other modules to 0 in the .fst file), I am getting ED 1st flapwise bending-mode DOF of blade as 0.5386Hz and ED 1st edgewise bending-mode DOF of blade as 0.737Hz.

Is there a document to which I can compare the coupled full system frequencies of  IEA-15-240-RWT-UMaineSemi floating machine (platform, tower, blade), if I try to identify the natural freq by following the openfast linearization process (runcampbell.m script, and then manually identifying the modes)?

ptrbortolotti, 2025-04-28:
Hi @Dellavar, your numbers look reasonable. I've recently run the linearization of the "onshore" model at 0rpm, with no gravity and no rotor tilt and modeling the blades in BeamDyn, and I obtained 0.506 Hz for blade flap and 0.693 Hz for blade edge. You can see more results here https://iopscience.iop.org/article/10.1088/1742-6596/2767/2/022018/pdf
I have not run the linearization for the floating model in a while, but I vaguely remember that @luwang00 had to help me out to remove the gravity from the model and achieve an isotropic rotor. Another data point is that https://github.com/OpenFAST/acdc helps linearize the model in OpenFAST. Unfortunately beyond this we don't a database of truth for natural frequencies of the system, and the OpenFAST model has changed a lot since its tech report was released in 2020

ptrbortolotti, 2025-04-28:
I ran a quick simulation with the floating model and ElastoDyn
I see: 
- 1st tower FA 0.467 Hz
- 1st tower SS 0.48 Hz
- 1st rotor flap 0.55 Hz
- 1st edge 0.75 Hz 

I use the OpenFAST toolbox

Here my files. Please double check them, it's possible that I have done something wrong

[floating.zip](https://github.com/user-attachments/files/19949588/floating.zip)

Dellavar, 2025-04-29:
Hi @ptrbortolotti ,
Thanks for the quick response and clarification.

I’ve tested the setup using the latest OpenFAST version along with the (https://github.com/OpenFAST/matlab-toolbox), enabling only the rotor, tower, and platform DOFs. I'm seeing the same results as you, and I also cross-checked with your provided files. Everything looks consistent, so I believe we're good to go.

Thanks again!

ptrbortolotti, 2025-04-29:
great, and just FYI I use the python toolbox https://github.com/OpenFAST/openfast_toolbox (it should be equivalent to the Matlab one)

closed 2025-04-29 by ptrbortolotti