IEAWindSystems/IEA-15-240-RWT #262: Differences between thrust in Beamdyn and Elastodyn

fnaos, 2025-05-13:
Hi! 

I am currently working with the version 1.1.3 of the openFAST model (v3.1.0) and I am observing significant differences between the aerodynamic thrust seen in my simulations when using Elastodyn and when using Beamdyn.

[image]

As it can be seen, the results are quite different, which made me wonder if there is any known bug for this version or any possible explanation for such divergence. In both simulations, the only thing that was changed in the input file, where the CompElast was adjusted for each simulation accordingly. Also, the DT had to be set to 0.0004s to run in the Beamdyn case. 

Giving you an overview of the simulations:

- Controller is deactivated.
- UMaine turbine with floater degrees of freedom deactivated and no hydrodynamic/mooring forces computed.
- Uniform wind of 10.59 m/s with a fixed rotor speed of 7.55 and fixed blade pitch of 1 degree
- Blade and tower degrees of freedom are set to TRUE
- In aerodyn, WakeMod = 2, AFAeroMod = 2 , DBEMT_Mod = 2, UAMod = 2

Any idea of what be causing such problem?

Cheers!

gbarter, 2025-05-13:
There have been a lot of bug fixes in BeamDyn since the 2022 OpenFAST v3.1 (see their [release notes](https://github.com/OpenFAST/openfast/releases)).  There have also been improvements in the reference turbine model since then.  I'm sure you have your own reasons for using that version, but we would be hard pressed to answer your question in technical detail without simply recommending to try it out using the latest releases.

ptrbortolotti, 2025-05-13:
something is clearly wrong. to debug, in addition to using a newer openfast, I would recommend deactivating the tower DOFs and switch to steady state aerodynamics

ptrbortolotti, 2025-05-13:
let me run a couple of checks. I'll get back to you

fnaos, 2025-05-14:
Thanks, Pietro. The problem of the differences in the Beamdyn and Elastodyn for the thrust forces are still intriguing me though.

[image]

I will also add the input files, in case it could be helpful

[IEA-15-240-RWT-UMaineSemi_AeroDyn15.txt](https://github.com/user-attachments/files/20204291/IEA-15-240-RWT-UMaineSemi_AeroDyn15.txt)

[IEA-15-240-RWT-UMaineSemi_ElastoDyn.txt](https://github.com/user-attachments/files/20204293/IEA-15-240-RWT-UMaineSemi_ElastoDyn.txt)

[IEA-15-240-RWT-UMaineSemi_Beamdyn.txt](https://github.com/user-attachments/files/20204306/IEA-15-240-RWT-UMaineSemi_Beamdyn.txt)

[IEA-15-240-RWT_InflowFile.txt](https://github.com/user-attachments/files/20204311/IEA-15-240-RWT_InflowFile.txt)

fnaos, 2025-06-17:
Perfect! Thanks, Pietro. It was something on my end

closed 2025-06-17 by ptrbortolotti