IEAWindSystems/IEA-15-240-RWT #273: Errors During UMaineSemi Simulation

capyfast, 2025-07-26:
Hello,

I am currently using OpenFAST v4.0.5 to run the IEA-15-240-RWT-UMaineSemi project and encountered several warnings related to HydroDyn and MoorDyn output channels.

Specifically, these HydroDyn warnings appear:
```
WARNING: The requested output channel is invalid: Wave1Elev  
WARNING: The requested output channel is invalid: WavesF1xi  
WARNING: The requested output channel is invalid: WavesF1zi  
... (and others similar)
```

Additionally, during MoorDyn initialization, I see the following:
```
Warning: invalid output specifier FX. Object ID missing.  
Warning: invalid output specifier FY. Object ID missing.  
Warning: invalid output specifier FZ. Object ID missing.
```

I suspect the MoorDyn warnings are caused by the output lines specifying just fx, fy, and fz without object IDs.

I am attaching the full simulation log for reference:  [sim_log.txt](https://github.com/user-attachments/files/21448356/sim_log.txt)

My questions are:
1. Should the HydroDyn output channels be updated or corrected in order to properly request available outputs?
2. What exactly do lines 56-58 specifying fx, fy, and fz in the MoorDyn input file represent? Should these lines be updated or removed?
3. Do these warnings impact simulation accuracy or are they simply related to output reporting?

Thank you very much for your guidance!

gbarter, 2025-07-27:
If you are using the latest release of the reference turbine OpenFAST files on the repo, then they are consistent with OpenFAST v4.1, which has some API changes relative to v4.0.  It will be tough to debug your Issue until we can ensure consistency on OpenFAST versioning.

capyfast, 2025-07-28:
I have run the simulation using the most recent version but still see the same errors, I am attaching the updated log

[validation_case.txt](https://github.com/user-attachments/files/21466919/validation_case.txt)

gbarter, 2025-07-28:
Ah, okay, thank you for taking the time to upgrade.  I double checked and those output channel names have indeed been updated to new values around OpenFAST v4.0 or just before that release.  I will make a note to change to the correct output channel names.  For now, you can safely ignore the warnings and proceed using the OpenFAST model.