IEAWindSystems/IEA-15-240-RWT #274: libdiscon.so could not be loaded

hec1122, 2025-08-07:
I am a beginner with OpenFAST and am attempting to run the IEA-15-240-RWT-UMaineSemi simulation via Windows PowerShell. However, I have encountered the following error:

The dynamic library .\../IEA-15-240-RWT/ServoData/libdiscon.so could not be loaded. Check that the file exists in the specified location and that it is compiled for 64-bit applications.

Details are as belows.
**************************************************************************************************
MoorDyn

Copyright (C) 2025 National Renewable Energy Laboratory
Copyright (C) 2025 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

   Parsing MoorDyn input file:
   C:\OPENFAST\openfast\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi\IEA-15-240-RWT-UM
   aineSemi_MoorDyn.dat
Warning: invalid output specifier FX.  Object ID missing.
Warning: invalid output specifier FY.  Object ID missing.
Warning: invalid output specifier FZ.  Object ID missing.
   Created mooring system:  3 lines, 6 points, 0 rods, 0 bodies.
   Finalizing initial conditions using dynamic relaxation.
   t=36.1  FairTen 1: 2.43899E+06, 2.43984E+06, 2.44053E+06
   Fairlead tensions converged to 0.1% after 36.1 seconds.
   Fairlead tension: 2.43899E+06
   Fairlead forces: -1.35268E+06, 0, -2.02951E+06
   Fairlead tension: 2.43902E+06
   Fairlead forces: 6.76355E+05, 1.17148E+06, -2.02953E+06
   Fairlead tension: 2.43902E+06
   Fairlead forces: 6.76355E+05, -1.17148E+06, -2.02953E+06
   MoorDyn initialization completed.
 Running ServoDyn.
 Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows).

FAST_InitializeAll:ED_Init:ED_ReadInput:ReadBladeInputs:ReadBladeFile:The ElastoDyn Blade file,
C:\OPENFAST\openfast\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi\../IEA-15-240-RWT/IE
A-15-240-RWT_ElastoDyn_blade.dat, DISTRIBUTED BLADE PROPERTIES table contains the PitchAxis
column.  This column is unused and will be removed in future releases
FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library
/home/runner/miniconda3/envs/test/lib/libdiscon.so could not be loaded. Check that the file
exists in the specified location and that it is compiled for 64-bit applications.

 OpenFAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

hec1122, 2025-08-08:
Thank you for your guidance, sir.
However, I am having difficulty understanding how to integrate the ROSCO repository correctly or adjust the file paths accordingly.
I have copied the `libdiscon.dll` file directly into the same path and modified the configuration line as shown below:
-------------------------------------------------------------------------------------------------------------------
"libdiscon.dll"  DLL_FileName - Name/location of the dynamic library {.dll [Windows] or .so [Linux]} in the Bladed-DLL format (-) [used only with Bladed Interface]
--------------------------------------------------------------------------------------------------------------------

However, another problem is occured. Could this be related to the file path ?
--------------------------------------------------------------------------------------------------------------------
 ROSCO:SetParameters:ReadControlParameterFileSub: >> A fatal error occurred when parsing data from "C:\OPENFAST\openfast\IEA-15-240-RWT-master (1)\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi\IEA-15-240-RWT-UMaineSemi_DISCON.IN".
 >> The variable "F_LPFType" was not assigned a valid value on line #8.

FAST_Solution0:CalcOutputs_And_SolveForInputs:SolveOption2:SrvD_CalcOutput:DLL_controller_call:ROS
CO:SetParameters:ReadControlParameterFileSub: >> A fatal error occurred when parsing data from
"C:\OPENFAST\openfast\IEA-15-240-RWT-master
(1)\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi\IEA-15-240-RWT-UMaineSemi_DISCON.IN
--------------------------------------------------------------------------------------------------------------------

dzalkind, 2025-08-11:
Hi,

Can you confirm you're using the latest version of the ROSCO library downloaded here? https://github.com/NREL/ROSCO/releases/tag/v2.9.0

There should be a short message at the start of the simulation.

Best, Dan